milesburton/Arduino-Temperature-Control-Library #184: MAX31850 limited to <256C with int16_t in DallasTemperature::calculateTemperature()

karlg100, 2020-12-27:
The MAX31850 supports a temp range of up to ~1600C.  This library truncates MSB with the use of int16_t in the calculateTemperature() function.

Addressed in #183

RobTillaart, 2020-12-28:
Interesting sensor to add. (Still need to read datasheet)
As it breaks the interface on some places e.g -127 constant I propose to change the version to 3.10.0 to indicate this. 

Q: do you intend to use multiple of these 31850s on one bus?

karlg100, 2020-12-28:
I do.  I have one application where I have 4. (PCB hasn't arrived yet).

On my original perf board rig, this library works out of the box and have used with 3 devices on a shared bus.

Have not tested that configuration however with my changes.

I can breadboard quickly two thermocouples to test my changes.  It should still work however as I didn't touch any of the device indexing portions of the library.

karlg100, 2020-12-29:
patch set complete and tested.  #183 

ready for review and merge.

closed 2022-06-03 by milesburton