milesburton/Arduino-Temperature-Control-Library #189: if A0-3 are not 1, will break resolution switch statement()

karlg100, 2020-12-29:
https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/798cfc217ad9b9fd3a0ecaae7613056f5cf890da/DallasTemperature.cpp#L351

It appears this switch statement will fail if any of the A0-3 bits are pulled down.

This is because bits 0-3 match these hardware set address bits.  Default they are pulled up.  But a user can configure these with hardware or dynamically pulling up or down.

The CONFIGURATION bit probably needs to be masked out with a | of 11110000 or 0xF0, then the macros redefined to be 0xN0.

Not sure what other repercussions this change would have however.

MalcolmBoura, 2021-01-31:
There is the same problem in setResolution(). It won't actually fail but it may write when it does not need to.
Needs a systematic look through all of the code.

MalcolmBoura, 2021-02-01:
The correct fix is to write a Scratchpad class which hides the details of how to access things such as the resolution. I have one in preparation.

closed 2022-06-03 by milesburton