milesburton/Arduino-Temperature-Control-Library #191: can't get DS18B20 with esp32 woring in PlatformIO while it is working out of the box in Arduino

cnoork, 2021-01-04:
I have tried many options like other gpio ports, different speed, older libraries and others but I am not getting control of the sensor in PlatformIO. See that there are many discussions about this sensor but no solution is working for me. What steps do I need to take to get it working as expected.

With kind regards,
Chris

RobTillaart, 2021-01-04:
ESP32 is 3.3V right?

milesburton, 2021-01-04:
Yes unless it has a reg iirc

cnoork, 2021-01-04:
Hi Rob, yes esp is 3.3V. The sensor will work from 3 to 5V and is already working for a while with a Arduino sketch on the same Wemos board

RobTillaart, 2021-01-04:
Can you measure the actual voltage?
Seen in the past that it was pretty close. 
Do you use a pull up resistor on the data line?  Think at 3.3V you can safely use 1K or 2K2

cnoork, 2021-01-04:
it is 3.28V and when I put the Wemos with the Arduino sketch on it is directly showing the correct temperature on my display. With the PlatformIO I get -127 Celsius. And hereby it does not matter to which pin I configure the onewire.

RobTillaart, 2021-01-04:
Have you tried a different lib e.g https://github.com/RobTillaart/DS18B20_RT - which is slightly simpler

Or try only Onewire - from https://www.avrfreaks.net/forum/ds18b20-only-onewire-library

```cpp
#include <OneWire.h>
int16_t dallas(int x,byte start){
    OneWire ds(x);
    byte i;
    byte data[2];
    int16_t result;
    do{
        ds.reset();
        ds.write(0xCC);
        ds.write(0xBE);
        for (int i = 0; i < 2; i++) data[i] = ds.read();
        result=(data[1]<<8) |data[0];
        result>>=4; if (data[1]&128) result |=61440;
        if (data[0]&8) ++result;
        ds.reset();
        ds.write(0xCC);
        ds.write(0x44, 1);
        if (start) delay(1000);
    } while (start--);
    return result;
}
void setup{
  dallas(A0,1);
}
void loop{
  float currentTemp = dallas(A0,0);
}
```
have very little time now, maybe tomorrow

RobTillaart, 2021-01-04:
> Do you use a pull up resistor on the data line? Think at 3.3V you can safely use 1K or 2K2

?

cnoork, 2021-01-05:
Hi, yes I have a pullup resistor, a 4,7k and is working. I have also not that much time now, I am will install Arduino again and will do some test again. I paused the project 1,5 year ago because of other projects. For the libraries I tried different onces and I think including you mentioned. In the mean while I have also asked the same question in the PlatformIO community because I do not know if it is the library or the PlatformIO it self which cause this.

RobTillaart, 2021-01-05:
These things happens, glad you got it working.  👍