milesburton/Arduino-Temperature-Control-Library #197: Four DS18B20's consistently about two degrees off?

jov58, 2021-03-04:
I'm trying to figure out why my DS18B20's are reporting too high temperatures.

To test I connected four DS19B20's to an ESP8266: two in a TO-92 case, one in a waterproof housing with a 50 cm cable and one waterproof with a 100 cm cable. I've checked them with [discover_fake_DS18B20 ](https://github.com/cpetrich/counterfeit_DS18B20), see annotations `fake?` and `genuine?` below.

To verify the temperature with other sensors, I also connected a DHT-11 (Sunfounder Humiture Sensor) and a DHT-22 (AM2302 from China). And placed an analog thermometer and a digital Daikin room thermostate near the breadboard.

These are the temperatures after a while, sampling every 5 seconds (DS18B20's in 11-bit resolution):

```
Daikin: 20.9  °C
Analog: 20.5  °C (about)
DHT-11: 22    °C
DHT-22: 21.7  °C
TO-92a: 23.00 °C Genuine? Also real Maxim Integrated according to supplier tinytronics.nl.
TO-92b: 22.94 °C Fake? Only returns 12-bit so not a DS18B20 but DS18S20?
W050cm: 22.88 °C Genuine?
W100cm: 22.63 °C Fake?
```

I don't blame the DHT-11 for being 1 degree off. The Chinese DHT-22 does the best job: only 0.8 off. But the 4 DS18B20's are consistently about 2 degrees off. If the circuit is switched on again in the morning after 10 hours, temperatures are immediately off by the same amount: DHT's about 1 degree, DS18B20's about 2 degrees.

Could this somehow be caused by the library? Could there be something else causing this? Or is it just a dramatic coincidence having four bad DS18B20's? Or, even worse, is it normal for DS18B20's to be 2 degress off? They should have a 0.5 degree accuracy according to the datasheet.

Any ideas?

RobTillaart, 2021-03-04:
Nice test,

First thought:: you assume that the Daikin and analog are correct

Please try the waterproof DS18 in boiling water (100°C)  and in meting ice
water.(0°C)  - [ I assume you are on sea level - 200mtr 1000hPa pressure
etc. }
DS18 can handle those temperatures so it is a good way to verify the sensor
accuracy.

R..

jov58, 2021-03-04:
> 
> First thought:: you assume that the Daikin and analog are correct
>
Yes. Because the Daikin is regulating the heating of our house (heat pump) and I think we'd feel it when the real temp would be almost 23C. The analog thermometer is a -30 to +30 one and reports fridge and freezer temperatures within 1C of the settings. A long way from 20C but still: always within 1C of the Daikin. So yes, I believe these two are closest to the real temperature.

Regarding 0 and 100C: good idea, I'll test that tonight.

BTW: connected my last genuine TO-92 Maxim: same 2 degrees. Sigh.

jaggzh, 2021-09-13:
Did you ever end up testing the 0 and 100C, @jov58 ?

closed 2022-06-03 by milesburton