milesburton/Arduino-Temperature-Control-Library #198: I'm getting -127 sometimes besides normal readings

spiutm, 2021-03-29:
I'm getting -127 sometimes besides normal readings, bad sensor?

12:43:46.525 ->   Temp = 21.63
12:43:46.902 ->   Temp = -127.00
12:43:48.078 ->   Temp = 21.44
12:43:49.296 ->   Temp = 21.44
12:43:50.515 ->   Temp = 21.38
12:43:51.640 ->   Temp = -127.00
12:43:52.109 ->   Temp = -127.00
12:43:53.328 ->   Temp = 21.38
12:43:54.545 ->   Temp = 21.38
12:43:55.762 ->   Temp = 21.38
12:43:56.984 ->   Temp = 21.38
12:43:57.638 ->   Temp = 21.38
12:43:58.858 ->   Temp = 21.38
12:44:00.076 ->   Temp = 21.38
12:44:01.249 ->   Temp = 21.31
12:44:02.465 ->   Temp = 21.31
12:44:03.120 ->   Temp = 21.31
12:44:04.339 ->   Temp = 21.25
12:44:05.554 ->   Temp = 21.25
12:44:06.767 ->   Temp = 21.31
12:44:07.985 ->   Temp = 21.25
12:44:08.686 ->   Temp = 21.25
12:44:09.905 ->   Temp = 21.25
12:44:11.123 ->   Temp = 21.31
12:44:12.345 ->   Temp = 21.25
12:44:13.563 ->   Temp = -127.00
12:44:14.081 ->   Temp = 21.25

milesburton, 2021-03-29:
You're polling quite frequently, do you have an adequate power supply? The
IC takes a moment to charge up the internal cap.

spiutm, 2021-03-29:
delay(1000); 
tested on a 12v and on usb pc power , same result

spiutm, 2021-03-29:
with delay 2000 same result
13:12:39.512 ->   Temp = 20.25
13:12:39.701 ->   Temp = 20.25
13:12:41.953 ->   Temp = 20.31
13:12:44.154 ->   Temp = 20.31
13:12:44.948 ->   Temp = 20.31
13:12:47.203 ->   Temp = 20.31
13:12:49.402 ->   Temp = 20.38
13:12:50.247 ->   Temp = 20.38
13:12:52.498 ->   Temp = 20.31
13:12:54.701 ->   Temp = 20.31
13:12:55.449 ->   Temp = 20.31
13:12:57.654 ->   Temp = 20.38
13:12:59.904 ->   Temp = 20.31
13:13:00.746 ->   Temp = -127.00
13:13:02.952 ->   Temp = 20.38

milesburton, 2021-03-29:
Depending on resolution it can take up to 750ms to take a reading, to be
fair it should be working but I'd  be interested to see what happens if you
bump up the sample time to say 3 seconds. Do you see the same problem?

milesburton, 2021-03-29:
Interesting, I doubt this is the library.

-127 is the 'error' constant that the library will return if it was unable
to take a reading from the IC. Whilst not a fix, you could discard any -127
readings.

spiutm, 2021-03-29:
ill investigate further cuz running your code stand alone works fire, but adding blynk in same sketch seems to makes it -127

spiutm, 2021-03-29:
found the problem, it is the blynk sync, even thou it's on different pin

pike2k, 2022-02-06:
Sorry to hijack your thread here. 
In my project I tried to add ESPAsyncWebserver and then I started getting a lot of -127 readings suddenly from my DS1820 sensor. When I removed ESPAsyncWebserver library completely the issues went away. Not sure the best way forward here...

RobTillaart, 2022-02-06:
How often do you refresh the webpage?
How often do you read the sensor?
Do you use RTOS?

These kind of issues are often better discussed e.g. on the Arduino forum.

pike2k, 2022-02-07:
I use Platform IO and regular Arduino. I'm on ESP32.
The webpage was a static Hello World example.
I read the sensor everytime the main loop restarts, tried 2000ms and 50ms, no difference to the -127 issue.
My current working theory is that there's a conflict between OneWire and ESPAsyncWebServer.

RobTillaart, 2022-02-07:
Never tried but you might investigate replacing onewire with onewireng https://github.com/pstolarz/OneWireNg

No idea if this is easy or not.

closed 2022-06-03 by milesburton