milesburton/Arduino-Temperature-Control-Library #204: yield() function causing problems uploading

Sandoxus, 2021-06-16:
When trying to upload from some devices you may get an error saying yield() is not declared in this scope. All you need to do is go into your DallasTemperature.h and DallasTemperature.cpp and delete where the yield() function is being called. After doing this go back into like 443 where there is an if statement with a nested while loop saying

void DallasTemperature::blockTillConversionComplete(uint8_t bitResolution) {

  if (checkForConversion && !parasite) {

    unsigned long start = millis();

    while (!isConversionComplete() && (millis() - start < MAX_CONVERSION_TIMEOUT ))

    delay(250); (ADD THIS IN PLACE OF yield())

  } else {

    unsigned long delms = millisToWaitForConversion(bitResolution);

    activateExternalPullup();

    delay(delms);

    deactivateExternalPullup();

  }

}

and basically you just add the delay(250); in place of the yield(); function and it will work perfectly

Sandoxus, 2021-06-16:
[DallasTemperature .h.cpp files.zip](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/6664570/DallasTemperature.h.cpp.files.zip)

RobTillaart, 2021-06-16:
The **yield()** function is not a delay function, although in practice it can give a delay in execution.

It is used in several RTOS systems to give a signal to the OS that this is a good moment to swap tasks. E.g. an ESP32 and ESP8266 has its wifi and bluetooth on separate tasks and the current implementation (I am told) needs attention on average every 5 ms to prevent overflow of some buffers or so. **yield()** takes care of that. 

An UNO has no RTOS but implements **yield()** as an empty function  in **hooks.c** 
```cpp
static void __empty() {
	// Empty
}
void yield(void) __attribute__ ((weak, alias("__empty")));
```

So platforms should better implement it somewhere in their global includes as **yield()** will show up in more and more libraries. 

maybe it is better to include a similar file in your platform.....
BTW what is your platform that does not support yield?

RobTillaart, 2021-06-16:
remark 2:

```cpp
if (checkForConversion && !parasite) {
  unsigned long start = millis();
  while (!isConversionComplete() && (millis() - start < MAX_CONVERSION_TIMEOUT ))
   //  just remove yield()
} else {
  unsigned long delms = millisToWaitForConversion(bitResolution);
  activateExternalPullup();
  delay(delms);
  deactivateExternalPullup();
}
```

if you just remove **yield()** it will also work 
as the loop blocks already on the ```millis() - start < MAX_CONVERSION_TIMEOUT )``` part

in fact it would even be faster as it would react instantly after the conversion is complete.

Sandoxus, 2021-06-16:
I'm currently having a ton of issues uploading my IDE file to my arduino mega 2560 remotely through VNC viewer. I'm getting a problem that says yield() is not declared in this scope. So I am running on raspbian arm64. I've been working with it all day am I'm coming across a ton of issues. I'm able to upload the code perfectly fine from a windows machine, but my goal is to be able to upload code remotely from the raspberry pi.

RobTillaart, 2021-06-16:
(never worked with the IDE on a RPI)
it sounds like the terminal installation misses dependencies. 
Good to hear you tackled the issue by installing the arm32/64 ide. 

Question: did you remove the terminal installation first, or did you copy the arm32/64 ide over the terminal installation?

Sandoxus, 2021-06-26:
I removed the Arduino IDE through the terminal by doing sudo apt remove arduino. Then I downloaded the deb package from the arduino website and installed it using dpkg and install. It updated me to a much newer version and also created the correct file in the Documents folder therefore, fixing my problem.

closed 2022-06-03 by milesburton