milesburton/Arduino-Temperature-Control-Library #205: Arduino Library spec rev.2.2

argandas, 2021-06-18:
This library is not compliant with Library Spec rev.2.2

```
------------------------------------------------------------------------------------------
# Action: Arduino CLI - Library install: "https://github.com/milesburton/Arduino-Temperature-Control-Library"
$ arduino-cli lib install --git-url https://github.com/milesburton/Arduino-Temperature-Control-Library
------------------------------------------------------------------------------------------
--git-url and --zip-path flags allow installing untrusted files, use it at your own risk.
Enumerating objects: 148, done.
Counting objects: 100% (148/148), done.
Compressing objects: 100% (87/87), done.
Total 148 (delta 52), reused 109 (delta 30), pack-reused 0
Error installing Git Library: library is not valid: missing header file "Arduino-Temperature-Control-Library.h"
Build failed: exit status 1
```

Source code requires renaming and moving to "src" folder

argandas, 2021-06-18:
Using Arduino CLI v0.18+

RobTillaart, 2021-06-19:
This is a known fact, therefor I added Github actions some weeks ago to verify the compliance 

- json check ==> for platform io usage
- arduino-test-runner => to run unit tests(*) and to test compilation of examples 
- arduino-lint => to check compliance with the Arduino library manager

These tests are available in https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/203
which needs to be merged into the master. 

In terms of compliance there are different levels, and the error messages you mention are generated on the most strict level as far as I know.  Using the **src** folder is only mandatory if the library has different code for different platforms, which is not the case.  About the naming, this library has a long history, way before the compliance checks and was renamed already in the past. Given it is well known in the community I would not rename it (there are just too many examples on the net that refer to this name)

(*) unit test are not made by me. There is just to little code that can be tested without a (full) simulation of the protocol which is mainly **oneWire** calls. I prefer testing with real sensors . 

(I am not the owner disclaimer) 
The library is an open source library so you may fork it and create a derived library as long as you operate within the license.

closed 2022-06-03 by milesburton