milesburton/Arduino-Temperature-Control-Library #206: Fails to compile with Arduino-ESP32 v2.0.0 Alpha

JohnMacrae, 2021-06-23:
Hi,

Just a heads up, the Simple example no longer compiles when used with this alpha. 

Error log:

In file included from /home/john/Arduino/libraries/OneWire/OneWire.cpp:144:
/home/john/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
/home/john/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/home/john/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
/home/john/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
/home/john/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
/home/john/Arduino/libraries/OneWire/util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc

Hope this helps and thanks for your support.

RobTillaart, 2021-06-23:
From the error messages it is clear that it is a problem with the onewire library. So it should be reported also there.

This library just builds on top of it, so cannot solve it as far as I can see.

pstolarz, 2021-08-03:
@JohnMacrae Could you check if [this fork](https://github.com/pstolarz/Arduino-Temperature-Control-Library) works fine for you?

JohnMacrae, 2021-08-17:
Works perfectly - many thanks :)

Kind Regards
John

closed 2022-06-03 by milesburton