milesburton/Arduino-Temperature-Control-Library #209: how do I set the temperature as an integer

Chipreader7, 2021-11-14:
Hi, I am brand new at this stuff.  I want to turn a fan on above a certain temperature, and a heater to turn on below a temperature.  I've just read Learn Electronics with Arduino, and in that book they have you set "int" for push button states and stuff.  Then you can turn on or off other items after you do a digitalRead to calculate the state of the "int".  

I can't figure out how to have the Arduino read the temperature and then act in response to it.  All the code examples I see simply send the temperature to the computer monitor.

RobTillaart, 2021-11-14:
@Chipreader7 

Several ways:
assume the variable temperature is a float that contains the last reading

1) assign the temperature to an integer variable and print that.
```cpp
int myTemp = temperature;    // just cut of the decimal part
Serial.println(myTemp);
```

2) do a cast while printing
```cpp
Serial.println( (int) temperature );   // just cut of the decimal part
```

3) print the float with 0 decimals
```cpp
Serial.println(temperature, 0 );   // does rounding
```

4) use round() function
```cpp
Serial.println( round(temperature) );    // explicit rounding
```
Note you could have used the round() function also when assigning it to an integer variable.

Normal this kind of questions should be asked on the Arduino.cc forum 
Issues here are those that are specific for this library.
You may close the issue

milesburton, 2022-06-03:
Thanks once again Rob

closed 2022-06-03 by milesburton