milesburton/Arduino-Temperature-Control-Library #210: Timingproblem DS18S20

PeterDDPain, 2021-12-14:
Hello there,

thank you for your nice Lib. Makes working with this sensor very handy!

I experienced a problem but maybe it is not a problem and i do something wrong.

I have a esp32 and the latest librarys from OneWire and DallasTemperature. But it doesent mater wich resolution is set, the conversion time is almost the same. I used the timing example with an arduino uno and with a esp32. I post here the different timings:

**_TIMING ARDUINO UNO_**

Test takes about 30 seconds for 4 resolutions
RES	TIME	ACTUAL	GAIN
9	94.00	751.10	799.0%
10	188.00	751.00	399.5%
11	375.00	751.00	200.3%
12	750.00	751.10	100.1%

**_TIMING ESP32_**

Test takes about 30 seconds for 4 resolutions
RES	TIME	ACTUAL	GAIN
9	94.00	496.95	528.7%
10	188.00	496.90	264.3%
11	375.00	496.95	132.5%
12	750.00	496.95	66.3%

Is there an issue with the lib or am I the issue by myself?  :D

Thanks in advance
Greetings

RobTillaart, 2021-12-14:
There is also the option of fake DS18B20's - https://github.com/cpetrich/counterfeit_DS18B20

PeterDDPain, 2021-12-14:
Ahhh....well the test said it is propably a counterfeit version.

Thank you a lot!!

RobTillaart, 2021-12-14:
You may close the issue :)

closed 2021-12-14 by PeterDDPain