milesburton/Arduino-Temperature-Control-Library #211: Additional documentation not available since milesburton.com is down

tysonmatanich, 2022-02-08:
It would be nice if the additional documentation could be hosted on GitHub instead of an external personal site.

Return 522 connection timed out
https://www.milesburton.com/Dallas_Temperature_Control_Library

RobTillaart, 2022-02-08:
What kind of info are you looking for?
Most information is in the datasheet.

tysonmatanich, 2022-02-08:
Not exactly sure. README.md states:
> Additional documentation may be found here https://www.milesburton.com/Dallas_Temperature_Control_Library

I was able to find a copy using the wayback machine if others find it useful (just reading through now myself)
https://web.archive.org/web/20210309025235/https://www.milesburton.com/Dallas_Temperature_Control_Library

hllhll, 2022-02-16:
API Description is missing; I.E. "How to use the library"
Yeah there are examples but reference is always needed

milesburton, 2022-06-03:
Servers are back online. Most if not all information should be contained within the library and also widely publicised. Let me know if this is not the case. Thanks all.

closed 2022-06-03 by milesburton