milesburton/Arduino-Temperature-Control-Library #212: Value of temperature changing when restart ESP32

Layochi, 2022-02-13:
Hello,
I'm getting an issue when I restart my ESP32 to read temperatures.
Temperatures that are readed by the DS18B20 after the restart are different (-0.5°C) from those read before...

I'm running an ESP32 24/24 7/7 and every hour, it reads the outdoor temperature and it sends it to a database. But every day, at 7:30 am I restart it and i'm getting a drop in temp as it's shown in the picture :

[image]

To read the temperature i use `tempSensor.getTempCByIndex(0);`
To restart the ESP32 i use : `ESP.restart();`

RobTillaart, 2022-02-14:
Strange effect, interesting ..
Some questions pop up
- how long does the restart take? (Milliseconds i assume)
- Are the sensors using parasite power?
- is the temp sensor near the processor or near other heat producing devices / sensors / cables?
-  has it enough free moving air around?

Layochi, 2022-02-14:
>  how long does the restart take? (Milliseconds i assume)

I don't know exactly but I suppose that it takes milliseconds.

> Are the sensors using parasite power?

The DS18B20 is powered by an external 5V power supply that works perfectly fine all day long so...

> is the temp sensor near the processor or near other heat producing devices / sensors / cables?

It's approximatly 1 meter away from all the electronics

> has it enough free moving air around?

The sensor is under water with a constant water flow arround it : the environment arround it is exactly the same before and after the restart.

RobTillaart, 2022-02-14:
OK,
I agree it is not expected that the water temperature changes so quickly.

The library is stable for many years, and not producing an error (like -127), so I would like to understand your setup in more detail.

I assume you sample the temperature with 12 bits (looking at the points in de graph)

Questions
- can you make a (simple) drawing of the setup? (photo?)
- Are there other devices in the setup?
- If so how are these powered?
- How is the ESP32 powered?

- can you modify the code so it makes a measurement just **before** the restart and one **after** restart?
Just add a measurement in setup() should do the trick. 
This will reduce the interval of the cause really around the restart.
- Have you tried if the same happens if you restart at e.g. 10:00 instead of 7:30
- Can you extract e.g. the last 100 temperature measurements from the database and post them here?
(preferred attached as a zip or text file; with 3 decimals if possible)
- Is it possible to post your code? (preferably minimized, remove the password etc)

I know a lot of questions, but it helps to get an idea of how the project looks like.

Layochi, 2022-02-14:
So that's my mont
[image]

Layochi, 2022-02-14:
So that's my setup (only a drawing because electronics is screwed in a tiny local) :
[image]

They're also relays, a light bulb (but it's turned off while reading temp), a very powerfull pump that's often turning to keep water clean but it's far away (behind the wall) from the DS18B20 and it's not running at night (8pm to 8 am) so it's not running at 7:30am...

The power supplies are very powerful so I don't think that they're the problem.

I just made few adjustements to send the two temperatures (one just before, one just after) and set the restart for 10 o'clock. I'lll keep you informed.

For measurments, I only have 48 of them (with two decimals...) because I keep only 2 days (48 hours) of measures and after that I do an average for the day (temperature history). 
[tempHistory.txt](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/8063717/tempHistory.txt)

Now my code : 
I use the esp32 as a web server and it's only responsing to request that are sended by a node.js server that runs commands everyday to read and format data.

OTA is also setup to provide easier acces to the code even if the ESP32 is in the local and Homekit to include SIRI (from IOS) to answer questions like "what's the temperature blablabla"...

I've had a button to take control of the pump but it "sinked" and is not working anymore (but it's not linked to the problem) so I just commented the few lines that were useful for it.

```#include "wifi_info.h" //Wifi config
#include "OTA.h" //OTA config
#include "homekit.h" //Homekit config

#include <WebServer.h>
#include <OneWire.h>
#include <DallasTemperature.h>

#define tempSensorPin 16
#define buttonPin 17
#define lightPin 18
#define pumpPin 19

OneWire oneWire(tempSensorPin);
DallasTemperature tempSensor(&oneWire);

int lastLight, lastPump, lastButton;

WebServer server(80);

void setup() {
  Serial.begin(115200);

  tempSensor.begin();

  pinMode(lightPin, OUTPUT);
  digitalWrite(lightPin, LOW);

  pinMode(pumpPin, OUTPUT);
  digitalWrite(pumpPin, HIGH);

  pinMode(buttonPin, INPUT_PULLUP);

  wifi_connect();
  setupOTA();
  my_homekit_setup();

  server.on("/pumpOn", pumpOn);
  server.on("/pumpOff", pumpOff);
  server.on("/lightOn", lightOn);
  server.on("/lightOff", lightOff);
  server.on("/restart", []() {                                                                 //HERE IS WHERE THE "MAGIC" APPEARS !
    server.send(200, "text/plain", "The ESP will restart...");
    delay(1000);
    ESP.restart();
  });
  server.on("/", handleRoot);
  server.onNotFound(handleNotFound);

  server.begin();

  debugln("WebServer has started !");
}

void loop() {
  handleOTA();
  server.handleClient();  //Wait for commands

  if (WiFi.status() != WL_CONNECTED) {
    WiFi.disconnect(true);
    wifi_connect();
  }

  if (lastPump != digitalRead(pumpPin)) {
    lastPump = digitalRead(pumpPin);
    debug("Pump State : ");
    debugln(lastPump ? "HIGH" : "LOW");
    homekit_pump_report((bool) lastPump);
  }

  if (lastLight != digitalRead(lightPin)) {
    lastLight = digitalRead(lightPin);
    debug("Light State : ");
    debugln(lastLight ? "HIGH" : "LOW");
    homekit_light_report((bool) lastLight);
  }

  /*if (digitalRead(buttonPin) != lastButton) {
    debugln("ButtonEvent triggered !");
    lastButton = digitalRead(buttonPin);
    if (!lastButton) digitalWrite(pumpPin, !digitalRead(pumpPin));
  }*/
}

void handleRoot() {
  tempSensor.requestTemperatures();
  float temp = tempSensor.getTempCByIndex(0);
  String header = "{\"pump\":" + String(digitalRead(pumpPin) ? "true" : "false") + ",\"light\":" + String(digitalRead(lightPin) ? "true" : "false") + ",\"temp\":" + String(temp) + "}";
  server.send(200, "text/plain", header);
  homekit_temp_report(temp);
  debugln("Root Request");
}

void pumpOn() {
  digitalWrite(pumpPin, HIGH);
  debugln("pumpOn Request");
  server.send(200, "text/plain", "200: Pump is ON");
}

void pumpOff() {
  digitalWrite(pumpPin, LOW);
  debugln("pumpOff Request");
  server.send(200, "text/plain", "200: Pump is OFF");
}

void lightOn() {
  digitalWrite(lightPin, HIGH);
  debugln("lightOn Request");
  server.send(200, "text/plain", "200: Light is ON");
}

void lightOff() {
  digitalWrite(lightPin, LOW);
  debugln("lightOff Request");
  server.send(200, "text/plain", "200: Light is OFF");
}

void handleNotFound() { // Page Not found
  debugln("Not found Request");
  server.send(404, "text/plain", "404: Not found");
}```

Layochi, 2022-02-14:
Today it was so strange look a that :

[image]

I don't know what happened !
It looks like it always add or remove 0.5°C...

RobTillaart, 2022-02-14:
Schema look good

----
> The power supplies are very powerful so I don't think that they're the problem.

Do you have a multimeter to verify?

----
> I just made few adjustements to send the two temperatures (one just before, one just after) and set the restart for 10 o'clock. I'lll keep you informed.

OK

----
> For measurments, I only have 48 of them....

The drop in the temperature seems to be longer than just one measurement after the reboot. (can you confirm?)
So it looks like it happens at restart but takes "some time to recover"

---- 
> I've had a button to take control of the pump but it "sinked" and is not working anymore (but it's not linked to the problem) so I just commented the few lines that were useful for it.

If the pump is not used you might need to skip these two lines in **setup()**.
```cpp
  pinMode(pumpPin, OUTPUT);
  digitalWrite(pumpPin, HIGH);
```

**Hypothesis**
These might draw extra current when the ESP32 starts, because of the 'broken' pump, which might affect the power supply?
Don't know but as the pump is not working one should comment all the lines referring to the pump pin.

----
Code looks good and straightforward for the rest.

RobTillaart, 2022-02-14:
> Today it was so strange look at that :
> 
> [image]
> 
> I don't know what happened ! It looks like it always add or remove 0.5°C...

Can you explain the two lines of the graph?

Layochi, 2022-02-14:
> The drop in the temperature seems to be longer than just one measurement after the reboot. (can you confirm?)
So it looks like it happens at restart but takes "some time to recover"

Yep, it seems to be longer than an hour.

> Do you have a multimeter to verify?

I'm sure that it is not the power supplies because if it was so it won't last more than a minute and I would have the good values at 8am...

> Hypothesis
These might draw extra current when the ESP32 starts, because of the 'broken' pump, which might affect the power supply?
Don't know but as the pump is not working one should comment all the lines referring to the pump pin.

Ok I missexplained : the button sinked not the pump : pump is working perfectly well and it is not turning between 8pm to 8am (so it is not interfering with the measures that are took at night...)

> Can you explain the two lines of the graph?

In green it's the old measures (yesterday) and in blue the new ones (today). So if you follow this guideline : the measure at the right in blue is the one from this hour (the one in green, is the temperature 25 hours ago) and the measure in green at the left was the temperature 48 hours ago (the one in blue, is the temperature 24 hours ago).

Layochi, 2022-02-15:
Hi ! I got the results !

So first, I made the test that you want me to do. I restart the ESP32 at 10 am and I got this result:

[image]

This morning, I received the temperatures, one just before the restart and one just after: both are exactly the same.
It's clear that there is still this "wave" at 8 o'clock but not at all at 10. So the restart might not be the problem.

### **But, I thought about the problem this night and one hypothesis comes to mind:**

In my code, when the ESP Restart, in the `setup()` function, it's written that the pump starts with ` digitalWrite(pumpPin, HIGH);`. So everyday, the pump was starting at 7:30am and not 8am... So every day, the pump started at 7:30 and not at 8:00... So it "mixed" the water of the whole pool and therefore the water that stayed outside during the night. When the night was cold, the water that was out of the skimmer (inlet of the water to be filtered) was colder than the water that stayed in the skimmer (location of the DS18B20) for the night.

**This should explain why there is a drop in temperature at this time of day.**

To check this hypothesis, I will let the pump run all night and see what happens tomorrow: stay tuned!

RobTillaart, 2022-02-15:
So the restart in itself is not the problem. It triggers the pump and so your hypothesis makes more sense.
So the sensor has probably been right all the time.

This shows why it is important to describe the complete system in all it details when there is a problem. Things are often more complex.

I am very interesting to see results of the next test.

Neither is the sensor

RobTillaart, 2022-02-15:
You might consider a 2nd sensor to measure outside temperature...

Layochi, 2022-02-15:
> I am very interesting to see results of the next test.

I will keep you informed.

> You might consider a 2nd sensor to measure outside temperature...

I've already done that using a weather API, but it's not very accurate...

RobTillaart, 2022-02-17:
Any progress?

Layochi, 2022-02-17:
Yes ! It was that ! 

[image]

Thanks for the help !

RobTillaart, 2022-02-17:
Ok, you may close the issue then

Layochi, 2022-02-17:
Yep ! Thanks a lot !

closed 2022-02-17 by Layochi