milesburton/Arduino-Temperature-Control-Library #213: Continiuns reboots

HugoCar77, 2022-04-23:
Hi..!!
I'm new and I'm started wiith this things and I have had a problem

In the line 39 in the Simple Example 
   float tempC = sensors.getTempCByIndex(0);

I by error I have put
   float tempC = sensors.getTempC(0);

Arduino IDE 1.8.19 compile OK...and I can upload to the board
But I have continiuns reboots and I think that has broken the sensor

I have corrected the problem and a new sensor and all OK...the old always -127

Board ESP32 DevKit v1 
Sensor DS18B20 (TO-92) in ONE_WIRE_BUS 26 with a simple little board 

Is normal this issue?

Thanks

[https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Simple/Simple.ino](url)

milesburton, 2022-04-23:
Good day,

This sounds like a power supply issue rather than a bug in the library. I'd suggest raising a post on the Arduino forum for general support, I'm sure they can advise you as needed.

To avoid confusion I'll close this ticket, if you think this is in error please let us know.

Thanks,
Miles

closed 2022-04-23 by milesburton

HugoCar77, 2022-04-24:
Hi...!!!

I also thought that but... you can see it in this video
I can't test in arduino

https://mega.nz/file/fZNCSCiD#TAyoRiiUz4_XtIDZsvTDAjr6TeLwIP2fxh_O4P0N-uA
[https://mega.nz/rich-file.png]<https://mega.nz/file/fZNCSCiD#TAyoRiiUz4_XtIDZsvTDAjr6TeLwIP2fxh_O4P0N-uA>
86.64 MB file on MEGA<https://mega.nz/file/fZNCSCiD#TAyoRiiUz4_XtIDZsvTDAjr6TeLwIP2fxh_O4P0N-uA>
mega.nz

Thanks..!!