milesburton/Arduino-Temperature-Control-Library #214: Where to find a reference?

Pareidol, 2022-05-20:
In the keywords.txt there are a lot of functions and methods listed.
But where to find a reference for all that stuff.
Except the examples in the library I didn't find anything.

RobTillaart, 2022-05-21:
Afaik The library has no formal reference document. 

Besides the .h file that describes the interface, it contains a readme.md file, which has a link to a website with additional documentation. Furthermore there is the datasheet of the DS18B20.

There are websites that describe how the library /DS18B20 is used in projects. 

What specific information are you looking for?

Pareidol, 2022-05-21:
Im trying to make a projekt with 6 sensors. To know exactly which temperature is from which sensore, I want to get the temperature via the address. But all examples I found are using the index.

I first wanted to write a small programm to get the address of a connected sensor (I know there is a getaddress command, but no reference hw to use it).
And then I hope with the getTempC I can also use the address, similar to the index.

RobTillaart, 2022-05-21:
to get the address of a device you should do something like this

```cpp
//    FILE: getAddress.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: getAddress from a single DS18B20
//     URL: https://github.com/milesburton/Arduino-Temperature-Control-Library

#include <OneWire.h>
#include <DallasTemperature.h>

OneWire oneWire(7);     // pin that connects the DS18B20's

DallasTemperature sensors(&oneWire);

DeviceAddress thermometer;

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  Serial.println(__FILE__);

  sensors.begin();

  if (!sensors.getAddress(thermometer, 0))
  {
    Serial.println("No address.");
  }
  else
  {
    printAddress(thermometer);
    Serial.println();
  }
}

void printAddress(DeviceAddress addr)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (addr[i] < 16) Serial.print("0");
    Serial.print(addr[i], HEX);
  }
}

void loop()
{
}

// -- END OF FILE --
```

Compiles on UNO, but not tested with a real sensor => your job ;)

The address printed can be used in the large sketch.

RobTillaart, 2022-05-21:
**DeviceAddress ** is just a uint8_t array.

RobTillaart, 2022-05-21:
Maybe better, use **examples/oneWireSearch/oneWireSearch.ino**

It will generate the code for the addresses found. Forgotten that I wrote that one long ago.

RobTillaart, 2022-05-21:
if you have the addresses you should be able to do something like

```cpp
//    FILE: address_array.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: use address array of multiple DS18B20
//     URL: https://github.com/milesburton/Arduino-Temperature-Control-Library

#include <OneWire.h>
#include <DallasTemperature.h>

OneWire oneWire(7);     // pin that connects the DS18B20's

DallasTemperature sensors(&oneWire);

uint8_t addr[][8] =     // array that holds the addresses
{
  { 0x28, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 },
  { 0x28, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 },
  { 0x28, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 },
  { 0x28, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 },
  { 0x28, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 },
  { 0x28, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
};

void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println(__FILE__);

  sensors.begin();
  sensors.requestTemperatures();

  for (int i = 0; i < 6; i++)
  {
    printAddress(addr[i]);
    Serial.println("\t");
    Serial.println(sensors.getTempC(addr[i]), 2);    //  print with 2 decimals
  }
}

void printAddress(DeviceAddress addr)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (addr[i] < 16) Serial.print("0");
    Serial.print(addr[i], HEX);
  }
}

void loop()
{
}

// -- END OF FILE --
```

RobTillaart, 2022-05-24:
@Pareidol 
if solved you may close the issue

Pareidol, 2022-05-25:
Hi @RobTillaart,
thanks for your help, with that I was able to solve my project so far.
I'm not the best coder, so I needed to "re-engineer" your programm, to understand what it does.
I tried to simplify it for anyone like me, with less programming experience.

For anyone with the same request, here I'l pose my two programms:

1) get the address of a single sensor connected to your microcontroller, in a format that can be used in the next step:

```c++
//#######################################################################################################
// FILE: getAddress.ino
// AUTHOR: Pareidol
// VERSION: 2.0
// PURPOSE: get the address from a single DS18B20
// Date: 25.05.2022
//#######################################################################################################

// include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS_PIN 2

// setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS_PIN);

// pass the oneWire reference to Dallas Temperature
DallasTemperature sensors(&oneWire);

// arrays to hold device addresses
DeviceAddress sensors_array;

//#######################################################################################################
// setup
void setup()
{
// start serial port
  Serial.begin(9600);
  while (!Serial);

// start sensor library
  sensors.begin();

// locate devices on the bus
  Serial.print("Found ");
  Serial.print(sensors.getDeviceCount(), DEC);
  Serial.println(" devices.");

// check if sensors_array contains any address
  if (!sensors.getAddress(sensors_array, 0))
  {
    Serial.println("Unable to find address for device 0"); 
    Serial.println();
  }

// print the address of the first found sensor
  else
  {
    Serial.print("Device 0 address: ");

    for (uint8_t i = 0; i < 8; i++)
    {
      Serial.print("0x");
      if (sensors_array[i] < 16) Serial.print("0");
      Serial.print(sensors_array[i], HEX);
      if (i < 7) Serial.print(", ");
    }    

    Serial.println();
    Serial.println();
  }
}

//#######################################################################################################
// main function
void loop()
{
  // nothing needed here
}
```

2) get the temperatures of the sensors, connected to the 2Wire_Bus, by their adress and not their index:
```c++
//#######################################################################################################
// FILE: getTemp.ino
// AUTHOR: Pareidol
// VERSION: 1.0
// PURPOSE: get temperatures from different sensors by their address
// Date: 25.05.2022
//#######################################################################################################

// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS_PIN 2

// setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS_PIN); 

// pass the oneWire reference to Dallas Temperature
DallasTemperature sensors(&oneWire);

// define sensor addresses
uint8_t sensor_1 [8]= {0x28, 0x9E, 0xA9, 0x0D, 0x18, 0x20, 0x06, 0x92};
uint8_t sensor_2 [8]= {0x28, 0x83, 0x4D, 0x10, 0x18, 0x20, 0x06, 0xE3};
uint8_t sensor_3 [8]= {0x28, 0xF6, 0x69, 0x0F, 0x13, 0x21, 0x01, 0x53};

//#######################################################################################################
// setup
void setup()
{
// start serial port
  Serial.begin(9600);
  while (!Serial);

// start sensor library
  sensors.begin();

// request temperatures
  sensors.requestTemperatures();

  
// print sensor 1 data
  printSensorTemperature(sensor_1,"Sensor#1"); // argument 1 = sensor address variable, argument 2 = printed sensor name

// print sensor 2 data
  printSensorTemperature(sensor_2,"Sensor#2"); 

// print sensor 3 data  
  printSensorTemperature(sensor_3,"Sensor#3"); 
}

//#######################################################################################################
// main function
void loop()
{
  // nothing needed here
}

//#######################################################################################################
// function to print sensor temperature
void printSensorTemperature(DeviceAddress sensor_x, String sensor_name)
{
  printAddress(sensor_x);
  Serial.println();
  Serial.print("Temperature " + sensor_name + ": ");
  Serial.print(sensors.getTempC(sensor_x), 3);    //print with 3 decimals
  Serial.println("°C");
  Serial.println();
}

//#######################################################################################################
// function to print sensor address
void printAddress(DeviceAddress sensor_x)
{
  Serial.print("Device address: ");
    for (uint8_t i = 0; i < 8; i++)
    {
      Serial.print("0x");
      if (sensor_x[i] < 16) Serial.print("0");
      Serial.print(sensor_x[i], HEX);
      if (i < 7) Serial.print(", ");
    }    
}
```

####################################################################
My programms are based on @RobTillaart's template, so all the credits belong to him 🙏

closed 2022-05-25 by Pareidol