milesburton/Arduino-Temperature-Control-Library #215: Why the lib do not calibrate the DS18B20 sesor

pierrot10, 2022-06-01:
Hello

I leant to calibrate the sensors by taking a measure in the ice (rowLow), a measure in the boiled water (rawRange) with DS18B20 sensor. I also mesure the ice (referenceLow) and boiled eater (referenceRange) with an accurate temometer which resist to the boiled water.

Then the bellow function, adjust the temperature.

Why your library doe not do that? Or offer that option?

```
float ds18b20_corrected(float *temperature, float rawLow, float referenceLow, float rawRange, float referenceRange)
{
  //Return the calculate value following the calibration step
  float correctedValue;

  //RawRange = RawHigh - RawLow
  //ReferenceRange = ReferenceHigh - ReferenceLow

  correctedValue = (((*temperature - rawLow) * referenceRange) / rawRange) + referenceLow;
  return correctedValue;
}
```

RobTillaart, 2022-06-01:
Good question.
There are several thoughts about calibration made in the past, I do not recall them all but here are my thoughts at the moment.
- The DS18B20 is factory calibrated which means that they should be pretty accurate
-The calibration function you propose is a linear mapping. 
  - It is relative simple so any user can do that.
- If you really want to calibrate a temperature sensor you need more
  - the error curve is seldom linear over the full range so
  - more measure points at least every 10° C, better every 5° C including below zero and above 100°
  - Pressure for measuring should be1013 mBar 
  - Probably you need a calibrated reference temperature sensor (might be sensirion)

If the function would be in the library it should calculate the correction factors once, so the performance is optimal.
```cpp
void setCalibration (float low , float high, float range, ...);
{
  // calculate _A and _B
}

float correct (float temperature)
{
  return temperature * _A + _B;
}

RobTillaart, 2022-06-02:
@pierrot10 
If there are no further questions please close the issue

ecosensors, 2022-06-02:
`Pressure for measuring should be1013 mBar`

Are you sure about that?
If you measure the low,high,rangeLow, rangeHigh at the same altitude and close to  the location where the DS18B20 will work, the pressure should not be considered? Isn't?

RobTillaart, 2022-06-03:
Depends on how you define calibration.

milesburton, 2022-06-03:
Per Rob's comments, please reopen if you feel this requires more investigation

closed 2022-06-03 by milesburton