milesburton/Arduino-Temperature-Control-Library #219: DEVICE_DISCONNECTED_C as -255 exceeds int8_t storage

bobwolff68, 2022-06-07:
DEVICE_DISCONNECTED_C was defined as -127 which has no issues around storage size. However, a change to make this -255 causes warnings and a compiler 'conversion' of -255 ==> 1.

This change has created warnings in what was a totally clean build library. And it may also cause real problems in the disconnected case.
int8_t DallasTemperature::getHighAlarmTemp(const uint8_t* deviceAddress)
This function returns an 8-bit signed value who's range is -127 to 128 but the DEVICE_DISCONNECTED_C is now -255 (used to be -127).

RobTillaart, 2022-06-07:
See commit - https://github.com/milesburton/Arduino-Temperature-Control-Library/commit/ac1eb7f56e3894e855edc3353be4bde4aa838d41#commitcomment-75490966

RobTillaart, 2022-06-07:
(from the commit comments)

Some options

1. revert the values and accept that they are in range of the MAX31850
2. create a separate class for MAX31850
3. create a derived class for MAX31850
4. replace all int8_t code into int16_t in the library.

Option 1 is the quick fix.

Drawback is that -127°C is within the MAX31850 range. Applications that use these low temperatures (liquid N2 ?) would possibly be affected, however:

According to the datasheet - https://datasheets.maximintegrated.com/en/ds/MAX31850-MAX31851.pdf - 

_The converters resolve temperatures to 0.25°C, allow readings
as high as +1768°C and as low as **-270°C**, and exhibit
thermocouple accuracy of 8 LSBs (2°C) for temperatures
ranging from -200°C to +700°C_

So even the current **DEVICE_DISCONNECTED_C   -255** is technically in the range of the MAX31850 converter.

milesburton, 2022-06-07:
> ID: <milesburton/Arduino-Temperature-Control-Library/issues/219/1148863258
> @github.com>

We could throw in an Option 5 and use a C++ Macro to enable those who need
such a broad range of temperatures?

I presume those who need -200C are likely outliers.

RobTillaart, 2022-06-07:
A macro sounds like a better fix than reverting. +1

bobwolff68, 2022-07-05:
Hey guys - Any chance of getting a decision on the solution here and getting it published? I'd love to not fork to get this settled in my own builds. Thanks!

milesburton, 2022-07-05:
I think a macro is the way to go, unfortunately I don't have an IC to actually test the change so I'm hesitant to raise a PR until I know it won't nuke anything.  

If you want to make the change Rob and I can give you a review?

bobwolff68, 2022-07-05:
Well, for me it's not a quandry - I'd just pick option 1 and revert the change. :)   But if you're going to cope with this otherwise, I'd think the 16-bit change would be proper but I'm not clear on whether a 2-byte value would be problematic for other reasons. But I'm concerned that the macro would map things that are technically still out of bounds. That seems like just askin' for trouble down the line. Right?

milesburton, 2022-07-05:
> <milesburton/Arduino-Temperature-Control-Library/issues/219/1174656800@

Let me grab a branch and I'll get back to you

milesburton, 2022-07-05:
Can someone sanity check this reversion: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/223

milesburton, 2022-07-05:
Take two. Helps if I don't mix up my data types: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/224

milesburton, 2022-07-05:
@bobwolff68 Change merged. Let me know if that solves your problem. I'll close the issue for now (we'll revisit the Macro when I get a test setup on the bench)

closed 2022-07-05 by milesburton

bobwolff68, 2022-07-05:
> <milesburton/Arduino-Temperature-Control-Library/issue/219/issue_event/6933610430

Thanks, Miles. Re-compiled with the change - zero warnings is a beautiful
thing. Let me know when the macro version comes aroun d- happy to give that
one a spin as well.

--bob