milesburton/Arduino-Temperature-Control-Library #225: DS18S20 does not work on parasitic power.

fdobrovolny, 2022-07-21:
When I run the DS18S20 in normal mode, everything works as expected with the 4k7 Ohm. When I try to run it in parasitic mode, the library does not detect the sensor. But when I manually ask for the power mode of the device via `sensors.readPowerSupply(sensorAddress)` I get True. When I try to read the temperature manually via `sensors.getTempC(sensorAddress)`, it returns -127. I tried to change the resistor to 5k Ohms and even tried the second constructor with a pullup, but nothing worked.

Any idea what to try next?

RobTillaart, 2022-07-22:
Did you try a smaller pull up e.g  1K?
That would give a higher current ..

isaul32, 2022-09-05:
Is it possible that the sensor is a counterfeit? At least DS18B20 has many. See https://github.com/cpetrich/counterfeit_DS18B20.

fdobrovolny, 2022-11-07:
As @isaul32 I found out it's a counterfeit sensor. Thank you for the help.

closed 2022-11-07 by fdobrovolny