milesburton/Arduino-Temperature-Control-Library #226: Disconnected sensor messes up sensor indices

janifr, 2022-08-01:
Disconnecting sensor 0 on 2 sensor system causes sensor 1 to return value -127 and sensor 0 return temperature of sensor 1 making data useless.

milesburton, 2022-08-01:
This is expected behavior. If you need consistency because you are
disconnecting ICs, you'll need to use the getByAddress function

janifr, 2022-08-01:
Thank you for super fast response! Could you please point me an example how to use the getByAddress function?

milesburton, 2022-08-01:
See line 51 and go from there: https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Multiple/Multiple.ino

closed 2022-08-01 by milesburton

janifr, 2022-08-01:
Thanks, problem solved. If anyone's interested, my code is here: https://github.com/janifr/esp8266_dallastemperature_aws_iot