milesburton/Arduino-Temperature-Control-Library #229: getAddress return the first address available even if not found the device

NilsRo, 2022-10-17:
I prepared 3 sensors but only 1 is connected. A call to getAddress is false but the last address found is written to the variable given. 

There should be no explicit address returned if the device is not found.

Reproduction: Call a device number that is not available with at least one device connected.

milesburton/DallasTemperature@^3.11.0

[image]
[image]

RobTillaart, 2022-10-18:
Can you post a minimal sketch that shows the problem?

The image just does not give information about the content of the variables before your call.

NilsRo, 2022-10-31:
Attached a simple sketch. 

Output with one DS18B20 connected:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
**289D8F16A8013C7A
289D8F16A8013C7A**

Output with now DS18B20 connected:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
**0000000000000000
0000000000000000**

Expected result in first example would be:
289D8F16A8013C7A
0000000000000000

[Dalles_Fehler.zip](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/9905281/Dalles_Fehler.zip)

closed 2025-01-13 by milesburton