milesburton/Arduino-Temperature-Control-Library #234: Update Licensing

iliketocode9, 2023-01-01:
# Licensing Update

I am very new to Github, so bear with me as I raise my first issue one here. 

I am interested in applying the code in this repository to a commercial setting and, after reading the closed issue about moving to an MIT license, I am wondering if that was officially done? The issue was closed some time ago, but readme.md was not updated to reflect any changes (that I can see anyway). Any verification would be greatly appreciated.

Best wishes!

milesburton, 2023-01-01:
Good day @iliketocode9.

The license has not changed, support from the key contributors (@RobTillaart) wasn't particularly strong, and I didn't receiving any comments advocating for or against. 

Personally I have no issue if this library is used in a commercial application (I'm 100% it already is). My only request is the developer contributes back to the code base, or the wider open source community.

iliketocode9, 2023-01-01:
Hello @milesburton. Thank you very much for the quick reply and the clarification. 

Of course, I completely understand. I respect the work and dedication of all the contributors. I will ensure that such is done.

milesburton, 2023-01-01:
No problem, best of luck with your project. Unless you have any further queries I'll close this ticket. Feel free to reopen if needed.

closed 2023-01-01 by milesburton