milesburton/Arduino-Temperature-Control-Library #245: ESP32-S2 not finding DS18B20

DrBwts, 2023-08-08:
Using an Adafruit QTPy ESP32-S2 and Arduino IDE 2.0.0,
DS18B20 non-parasitic, wired with 4.7K pull up on DQ
Compiles & uploads fine but just keep getting the `No more addresses.` message over & over.

RobTillaart, 2023-08-08:
Have you tried the example sketch (Simple one)?
Does it also give that output?

DrBwts, 2023-08-09:
Just ran the `Simple.ino` code and get,

`Requesting temperatures...DONE
Error: Could not read temperature data`

DrBwts, 2023-08-17:
Sorry for the delay, been on holiday.

Q1: It's all on a breadboard at the moment so no wires longer than 5cm
Q2: QtPy, PC & monitor are the only other things on the bench, no power cables near by.
Q3: Will try that, just realized I have no 1K on hand.
Q4: powered from the 5V output form the QtPy which in turn is powered via USB

RobTillaart, 2023-08-17:
> Q3: Will try that, just realized I have no 1K on hand.

A few 4K7 in parallel mode will lower the effective R too

freman, 2024-04-12:
I have been having weird problems on my esp82666 and I honestly put it down to 3.3v, I swapped pins off D0 to D15.

These are purchased from an extremely reputable supplier so I'm fairly sure they're genuine. Some times one would be found, sometimes all 5, sometimes only 4, often zero.

I have however found a work around that works very reliably for myself.

```cpp
	sensors.begin();  // Start up the library
	delay(100);
	sensors.begin();  // REStart up the library
```

RobTillaart, 2024-04-12:
Do you have a pull up on the data line?
That improves the signal quality.

freman, 2024-04-12:
I do a 4.7k, this isn't my first go at 1-Wire, I still have some original in sample packaging devices around here somewhere from back before redis was a thing :D.

Once I order my PCBs I'll be inclined to swap more around but I gave up entirely on the breadboard and I cbf swapping it on the prototyping board, I will also have new esp's coming with hopefully more stable 3.3v rails from a more reputable non-ebay supplier. I got around to plugging it into my scope and watching it.

I was just offering up my work around for the other fella to see if that helped his situation.

madmacks59, 2024-07-13:
I have been having basically the same problem with the sensors not starting. I thought it mught be a tmins isses after the "begin" so I added a "delay(500);" and that didn't solve anything. I then I added a "  Serial.print(sensors.getDeviceCount(), DEC);" after the sensors.begin() and things worked. I found this ticket and changed the print to another sensors.begin() and that worked as well. Wonder what is going on...

RobTillaart, 2024-07-14:
@madmacks59 
Cause is still not found (as far as I know), however your post might give a hint to the cause (initialization of the oneWire bus).

closed 2025-01-13 by milesburton