milesburton/Arduino-Temperature-Control-Library #246: Arduino nano esp32 not findiding

AngelRionCervi, 2023-08-29:
Using an arduino nano esp32, I always get -127 as a result with the `Simple.ino` example.
I also tried with some resistors, 1K and 4.7K.

No issues with a XIAO ESP32C3 or uno R4 WIFI.

RobTillaart, 2023-08-29:
(No experience with the NANO ESP32)
Which pins of the NANO did you use for data pin? 
Have you tried others?  
Some of the pins might have additional functions already causing problems.
(Had a pin problem with another ESP32 when trying some software SPI, so this could be similar).

AngelRionCervi, 2023-08-29:
Ok it seems I wrote this issue a bit too fast.

I used the the variable names (A0 for example) instead of the GPIO number (1 in this case).
If I use the GPIO numbers from the pinout diagram, it all works fine:
[image]

Thank you !

closed 2023-08-29 by AngelRionCervi