milesburton/Arduino-Temperature-Control-Library #250: Location for MAX31825

pmr1, 2024-02-09:
This version of  1-wire temperature sensors has an address function allowing up to 64 addresses. Each address specifies the temperature sensor can be location.  It does this by sampling the value of a resistor connected to ADD0 and ground  .. see the data sheet for specific values.

I have updated library files to include a getLocation function.  It samples the resistor value and writes the address to the 6  lsb bits in the status register.   To access this you have to read the scratchpad .. the address located at scratchpad[2]
[DallasTemperature.zip](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/14221493/DallasTemperature.zip)

RobTillaart, 2024-02-09:
Sounds interesting, I do not recall seeing this in the datasheet (which page?)

pmr1, 2024-02-09:
See page 13 on attached data sheet 
[max31825_281123.pdf](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/14221571/max31825_281123.pdf)

pmr1, 2024-02-09:
Yes here is an arduino script running on  an arduino mini pro  .. but that should not matter.  This outputs the temperatures from 5 sensors:   
row listing  from left to right, Device type, run, temperature 0 , location 0, temperature 1, location1,  temperature 2 location 2,  etc.
The device type, 0x3B is a MAX31825.

I only had one sensor attached here
[temp37.zip](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/14221679/temp37.zip)

A 10k resistor attached to ADD0 listed as having an address 0x3a

:3b ,0 ,  ,13.94,3a, ,-127.00,3a, ,-127.00,3a, ,-127.00,3a, ,-127.00,3a

RobTillaart, 2024-02-09:
Datasheet P13

#### Address
_Although the 64-bit ROM code allows each 1-Wire device on a bus to be identified for communication purposes, it does
not provide any information about the location of the device. The MAX31825 includes two address pins (ADD0 and
ADD1). ADD0 can be connected to an external resistor whose value is measured by the MAX31825 in response to the
Convert Location command, resulting in five location address bits (A4:A0) being stored in the **Status** register. Because
the location resistor values on the board are known, this location address allows the location of the MAX31825 to be
uniquely identified. Mapping of the address selection resistor value to **A4:A0** is shown in Table 1. In addition to ADD0, the
ADD1 input can be connected to GND or VDD (or DQ in parasite-power mode). This selects the value of bit **A5**, yielding
a total of 64 available addresses. A5 = 1 when ADD1 is connected to VDD and 0 when ADD1 is grounded._

- 5 bits A0..A4 are defined with the 1% resistor at AD0  ==> so less precise resistors could cause address clashes
- 6th bit A5 is defined by the AD1
- location is set in the status register of the ScratchPad.

RobTillaart, 2024-02-09:
@pmr1 
OK, I understand from the prototype code how one gets a location from the MAX31825.
The implementation has to be engineered, with a more clear API as a DS18B20 does not support this call

I would propose something like this

```cpp
int DallasTemperature::getLocation(const uint8_t* deviceAddress)
{
  if (_wire->reset() == 0) return -1;  //  not supported
  _wire->select(deviceAddress);
  _wire->write(DETECTADD);
  if (_wire->reset() == 0) return -1;

  readScratchPad(deviceAddress, ScratchPad);
  return ScratchPad[2];
}
```

RobTillaart, 2024-02-09:
Should add a test for model in the function
```cpp
#define DS1825MODEL  0x3B   // also MAX31825
```

pmr1, 2024-02-09:
Ok but the existing api does recognise the 31825 and it does have a get address function so no need ro define an extra constant.

RobTillaart, 2024-02-09:
That model I mentioned is already defined in the .h file (did not invent it)

What I do not know if the original DS1825, which has the same model byte as the MAX31825, does support location.

Another question is what happens if the 2 lines are not connected / floating?
What does getLocation() return?
Can you test?

pmr1, 2024-02-09:
1. No the Max 31825 has a different scratchpad to the DS1825.
2. if Add0, 1 are left connected temperature still reads.

pmr1, 2024-02-09:
Oh right, I see your point the DS1825 does have an address function but operates via external pins and the address is seen in the configuration register. The Max31825 comes in a solder bump package measuring 0.
8mm x 1mm and is a bugger to solder. So using a resistor is a convenient way of fixing an address.

closed 2025-01-13 by milesburton