milesburton/Arduino-Temperature-Control-Library #254: Slow fetching data even when using setWaitForConversion(false)

bobemoe, 2024-06-27:
I am reading 12+ sensors. I use `setWaitForConversion(false)` so `requestTemperatures()` returns instantly and as I understand it the conversions are done in the background. Then when I want the temps I use `getTempC(address)` which takes 13ms to return. This is much faster than waiting for conversion but still when reading 10 sensors this is 130ms which I cant afford to block for, and we are wanting to add more sensors...

I refactor my code to read only one 13ms block per loop witch spreads out the blocking rather than having it all at once but I would still like to investigate the reason for the delay. Is it purely the unavoidable transfer time of the converted data from sensor to processor? Or are there other ways to optimise this?

Since conversion has already been done at this point I assume lowering resolution will not have any impact so have not tested this. Is this correct?

By my (cough, ChatGPT) calculations:

> The data transfer time for one DS18B20 temperature sensor over the OneWire bus can be summarized as follows:
> 
>     Communication Speed: 16.3 kbps (kilobits per second)
>     Command Bits: 8 bits (1 byte) to send the "Read Scratchpad" command
>     Data Bits: 72 bits (9 bytes) to read the sensor's scratchpad
> 
> Total Bits Transferred
> 
>     Total Bits per Sensor: 8 bits (command) + 72 bits (data) = 80 bits
> 
> Time Calculation
> 
>     Time to Transfer 80 Bits:
>     Time = 80 bits / 16300 bits per second ≈ 0.0049 seconds
>     Time in Milliseconds: Approximately 4.9 milliseconds
> 
> This is the theoretical minimum time required to transfer the temperature data for one sensor, excluding any additional overhead from command initiation, library processing, or microcontroller handling.

So I'm guessing the additional 8ms are overheads and unavoidable? Or can we shave them off somehow?

Any tips or tricks or optimisations appreciated? Thanks

RobTillaart, 2024-06-27:
Interesting question,
Can you share your (stripped) code?

RobTillaart, 2024-06-27:
One thing that might be possible is to reduce the reading of the ScratchPad in the library.
For the temperature you only need the first two bytes.
[image]

RobTillaart, 2024-06-27:
Measurements with my library with an UNO, timing for getTempC only

if fetching 9 bytes it takes ~28 millis, if fetching 2 bytes it takes ~24 millis. 
so about ~4 millis to be gained when only fetching 2 bytes.

don't know how this gain ports to another processor.

bobemoe, 2024-12-10:
Thanks for confirming my thoughts and timings. Sorry for the slow reply.  I implemented the round robin approach. Here is my code in case anyone else is needing the same thing. Has been working great :)

```
#ifndef TEMPSENSORS_H
#define TEMPSENSORS_H

#include <OneWire.h>
#include <DallasTemperature.h>

class TempSensors {
public:
    TempSensors(uint8_t pin) : oneWire(pin), sensors(&oneWire), currentIndex(0), deviceCount(0) {}

    void begin() {
        sensors.begin();
        sensors.setWaitForConversion(false);
        deviceCount = sensors.getDeviceCount();
        for (int i = 0; i < deviceCount; i++) {
            sensors.getAddress(addresses[i], i);
        }

        sensors.requestTemperatures();
        updateAll();
    }

    void updateOne() {
        if (deviceCount == 0) return;

        temperatures[currentIndex] = sensors.getTempC(addresses[currentIndex]);
        currentIndex = (currentIndex + 1) % deviceCount;

        sensors.requestTemperatures();
    }

    void updateAll() {
        if (deviceCount == 0) return;

        for (int i = 0; i < deviceCount; i++) {
            temperatures[i] = sensors.getTempC(addresses[i]);
        }
        sensors.requestTemperatures();
    }

    float getTempC(const DeviceAddress& address) {
        for (int i = 0; i < deviceCount; i++) {
            if (memcmp(addresses[i], address, sizeof(DeviceAddress)) == 0) {
                return temperatures[i];
            }
        }
        return DEVICE_DISCONNECTED_C;
    }

    int getDeviceCount() const {
        return deviceCount;
    }

    bool getAddress(DeviceAddress& address, int index) const {
        if (index < 0 || index >= deviceCount) return false;
        memcpy(address, addresses[index], sizeof(DeviceAddress));
        return true;
    }

private:
    OneWire oneWire;
    DallasTemperature sensors;
    DeviceAddress addresses[32];
    float temperatures[32];
    int deviceCount;
    int currentIndex;
};

#endif // TEMPSENSORS_H

```

I use it like:

```
#include "src/TempSensors.h"

TempSensors tempSensors(PB1);

void setup() {
    tempSensors.begin();
}

void loop() {
    // update just one sensor per loop (~15ms)
    tempSensors.updateOne();
    
    // can access all temps at any time fetches from cache
    for (int i = 0; i < tempSensors.getDeviceCount(); ++i) {
        DeviceAddress address;
        tempSensors.getAddress(address, i);
        float temp = tempSensors.getTempC(address);
    }

}
```

closed 2024-12-10 by bobemoe