milesburton/Arduino-Temperature-Control-Library #265: DS18B20 is not responding on version 4.0.1

Chico0212, 2025-01-20:
The pinnout is right, and the sensor is responding properly when using 3.9.0, but, with 4.0.1 it's not working

milesburton, 2025-01-20:
Hey @Chico0212. Hmm can you share the details of your platform? Which IC are you using? I've had one other report of incorrect readings but nothing suggesting it's outright not working.

Chico0212, 2025-01-20:
> Hey [@Chico0212](https://github.com/Chico0212). Hmm can you share the details of your platform? Which IC are you using? I've had one other report of incorrect readings but nothing suggesting it's outright not working.

Hi my friend, thank you for the attention, check the original post, i edited it.

milesburton, 2025-01-20:
> > Hey [@Chico0212](https://github.com/Chico0212). Hmm can you share the details of your platform? Which IC are you using? I've had one other report of incorrect readings but nothing suggesting it's outright not working.
> 
> Hi my friend, thank you for the attention, check the original post, i edited it.
> 
> i am using an esp32 and arduino IDE

Hey buddy thanks for getting back to me. Is this running on an Arduino Uno or one of the other platforms like the ES8266?

Chico0212, 2025-01-20:
@milesburton esp32-wrover-e, it's on a lilygo dev kit (SIM 7670E with gps)

milesburton, 2025-01-20:
Hmm. I have a few ideas of what could be the cause but annoyingly I don't have one to hand. It could be a timing issue but I'd need to investigate. Thanks for raising this bug, for now stick with the previous version till we can figure this out.

milesburton, 2025-01-21:
Thanks @Chico0212 I think we've found the issue as Rob has highlighted. Would you like to try out the draft version of 4.0.2? See my PR here: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/267

RobTillaart, 2025-01-21:
@Chico0212 

Add a delay(750) in your code might solve the issue.

```cpp
#include "DallasTemperature.h"
#include "OneWire.h"

#define TEMP_PIN 32

OneWire oneWire(TEMP_PIN);
DallasTemperature sensors(&oneWire);
DeviceAddress endereco_temp;

void getTemperature() {
  sensors.requestTemperatures();

  delay(750);  // to give the sensors time to do the measurement.  <<<<<<<<<<<<<<<<

  Serial.print("Temperatura = ");
  Serial.println(sensors.getTempC(endereco_temp), 2);

  if (!sensors.getAddress(endereco_temp, 0)) {
    Serial.println("SENSOR NAO CONECTADO");
  } else {
    Serial.print("Temperatura = ");
    Serial.println(sensors.getTempC(endereco_temp), 1);
  }

  data.temperature = sensors.getTempC(endereco_temp);
}
```

milesburton, 2025-01-21:
This should now be reinstated with the latest release (4.0.2). Please let me know if this resolves your problem.

closed 2025-01-21 by milesburton

Chico0212, 2025-01-21:
> This should now be reinstated with the latest release (4.0.2). Please let me know if this resolves your problem.

huuum, this version is not available for me yet, but i'll wait for it, thanks so much

milesburton, 2025-01-21:
> ID: <milesburton/Arduino-Temperature-Control-Library/issues/265/2605451949
> @github.com>

It should appear soon. The Arduino platform polls Github, 'we' don't push
changes. We'll have to wait till they detect this repo has changed but it
shouldn't be long. You can always copy the changes directly to your local
if you want?

GalaxJann, 2025-03-31:
Hey, i had the same issue with the DS18B20. For those where the issue persists, try to add the following line in `void setup()` method:

`sensors.begin();` - Where `sensors` is your _DallasTemperature_ object.

This one fixed it for me, when i updated my _DallasTemperature_ library from **3.9.0** to **4.0.3**.