milesburton/Arduino-Temperature-Control-Library #266: After upgrading from 3.9.1 to 4.0.1 the readings from DS1820 have not been correct any more.

NorbIdeas, 2025-01-20:
After upgrading from 3.9.1 to 4.0.1 of this wonderful carefree appreciated library the readings from DS1820 have not been correct any more. But from the parallel wired DS18B20 very well. Both sensors had the same temperature. Each sensor employed alone did not change anything. The sketch runs on Arduino MKR ZERO (SAMD21 Cortex-M0+ 32bit low power ARM® MCU)

A copy from Serial Monitor of Arduino IDE 2.3.4 on Xubuntu 20.04.1
```

                     DS1820  DS18B20
"2025/1/18 22:28:56" 22.19°C 22.31   // v3.9.1
Logging to: LOGGER65.CSV             // compile and upload with the upgraded library
"2025/1/18 22:29:19" 2.81°C 22.31°C  // v4.0.1
"2025/1/18 22:29:29" 2.81°C 22.31°C  // v4.0.1

```

RobTillaart, 2025-01-20:
Disclaimer: no time to investigate on short term. 

So first sensor fails and second works?
Can you post your sketch?  or a minimal version that shows the problem?
Issue #265 also had a failing sensor, but this has a mix that might help to diagnose.

NorbIdeas, 2025-01-21:
```
   Serial.print(sensors.getTempCByIndex(0));
    Serial.print("°C");
    Serial.print(" ");
    Serial.print(sensors.getTempCByIndex(1));
    Serial.print("°C");

```
I had the setch well working with two parallel wired sensors, one DS1820 and one DS18B20. I got on the Serial Monitor of Arduino IDE output lines like this:
`"2025/1/18 22:28:56" 22.19°C 22.31°C`
Arduino IDE offered me to install a new version of Dallas Temperature library. I confirmed.
I knew: if I would compile and upload from now on, my sketch would be compiled with an new version of Dallas Temperature, and I had an eye on possible changes. 
I experimented with unplugging and plugging the sensors and noticed, that the reading of DS1820 resulted in 2.81 instead of 22.19

Increasing the sensors temperature from about 18°C to 30 °C increases the output from 2.69 to 3.50

**Experiment with library versions**
In Arduino IDE I step back with Manage Libraries from 4.0.1.to 3.9.0. During this I confirmed with "Install All" to a "dependency currently not installed" of Wire. 

```
"2025/1/21 10:24:37" 2.62°C 19.87°C
"2025/1/21 10:24:47" 2.62°C 19.87°C
"2025/1/21 10:24:57" 2.62°C 19.87°C   // 4.0.1
"2025/1/21 10:25:15" 21.06°C 19.87°C  // 3.9.0
"2025/1/21 10:25:25" 21.00°C 19.87°C
```

Then I stepped forward again to 4.0.1

```
"2025/1/21 10:27:25" 21.06°C 19.87°C
"2025/1/21 10:27:35" 21.00°C 19.87°C  // 3.9.0
"2025/1/21 10:27:52" 2.62°C 19.87°C   // 4.0.1
"2025/1/21 10:28:2" 2.62°C 19.87°C

```
Using two sensors of DS18B20 type, I don't see any changes.
I'm sorry, I don't have a second DS1820 sensor to rule out a unique behavior of this sensor.
Thanks

RobTillaart, 2025-01-21:
So in short: the 4.0.1 version of the library fails at converting the raw data to temperatures for the **DS1820** (without the B)

RobTillaart, 2025-01-21:
**calculateTemperature()** seems to have lost the DS1820 part.

Please replace the following function in the **DallasTemperature.cpp** file, line 526

```cpp
int32_t DallasTemperature::calculateTemperature(const uint8_t* deviceAddress, uint8_t* scratchPad) {
    int32_t fpTemperature = 0;

    // looking thru the spec sheets of all supported devices, bit 15 is always the signing bit
    int32_t neg = 0x0;
    if (scratchPad[TEMP_MSB] & 0x80)
        neg = 0xFFF80000;

    // detect MAX31850
    if (deviceAddress[0] == DS1825MODEL && scratchPad[CONFIGURATION] & 0x80) {
        if (scratchPad[TEMP_LSB] & 1) { // Fault Detected
            if (scratchPad[HIGH_ALARM_TEMP] & 1) {
                return DEVICE_FAULT_OPEN_RAW;
            } else if (scratchPad[HIGH_ALARM_TEMP] >> 1 & 1) {
                return DEVICE_FAULT_SHORTGND_RAW;
            } else if (scratchPad[HIGH_ALARM_TEMP] >> 2 & 1) {
                return DEVICE_FAULT_SHORTVDD_RAW;
            } else {
                return DEVICE_DISCONNECTED_RAW;
            }
        }
        // We must mask out bit 1 (reserved) and 0 (fault) on TEMP_LSB
        fpTemperature = (((int32_t)scratchPad[TEMP_MSB]) << 11)
                       | (((int32_t)scratchPad[TEMP_LSB] & 0xFC) << 3)
                       | neg;
    } else {
        fpTemperature = (((int16_t)scratchPad[TEMP_MSB]) << 11)
                       | (((int16_t)scratchPad[TEMP_LSB]) << 3)
                       | neg;
    }

  /*
   DS1820 and DS18S20 have a 9-bit temperature register.

   Resolutions greater than 9-bit can be calculated using the data from
   the temperature, and COUNT REMAIN and COUNT PER °C registers in the
   scratchpad.  The resolution of the calculation depends on the model.

   While the COUNT PER °C register is hard-wired to 16 (10h) in a
   DS18S20, it changes with temperature in DS1820.

   After reading the scratchpad, the TEMP_READ value is obtained by
   truncating the 0.5°C bit (bit 0) from the temperature data. The
   extended resolution temperature can then be calculated using the
   following equation:

                                    COUNT_PER_C - COUNT_REMAIN
   TEMPERATURE = TEMP_READ - 0.25 + --------------------------
                                           COUNT_PER_C

   Hagai Shatz simplified this to integer arithmetic for a 12 bits
   value for a DS18S20, and James Cameron added legacy DS1820 support.

   See - http://myarduinotoy.blogspot.co.uk/2013/02/12bit-result-from-ds18s20.html
   */
  
  if ((deviceAddress[DSROM_FAMILY] == DS18S20MODEL) && (scratchPad[COUNT_PER_C] != 0)) {
    fpTemperature = (((fpTemperature & 0xfff0) << 3) - 32
                    + (((scratchPad[COUNT_PER_C] - scratchPad[COUNT_REMAIN]) << 7)
                       / scratchPad[COUNT_PER_C])) | neg;
  }

    return fpTemperature;
}
```

milesburton, 2025-01-21:
I think you're right @RobTillaart.

milesburton, 2025-01-21:
We're also missing DSROM_FAMILY

milesburton, 2025-01-21:
This also needs to be added around line 26

// DSROM FIELDS
#define DSROM_FAMILY    0
#define DSROM_CRC       7

milesburton, 2025-01-21:
OK folks, I think I have a fix which is working locally. Let me know your thoughts https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/267

closed 2025-01-21 by NorbIdeas