milesburton/Arduino-Temperature-Control-Library #268: readme example need the delay(750) too

RobTillaart, 2025-01-21:
Just noticed it in the readme.md

```cpp
void loop(void) { 
  sensors.requestTemperatures(); 

  delay(750); 

  float tempC = sensors.getTempCByIndex(0);
  Serial.print("Temperature: ");
  Serial.print(tempC);
  Serial.println("°C");
  delay(1000);
}
```

milesburton, 2025-01-21:
Added to master. I'm about to raise another PR to add a new example so we can bump the version up if you agree it's useful

RobTillaart, 2025-01-21:
Sounds like a very practical example, which probably needs a small readme (e.g. change passwords or so).

milesburton, 2025-01-21:
> <milesburton/Arduino-Temperature-Control-Library/issues/268/2605581711@
> github.com>

I was also thinking the same. Not sure if that detail is best kept in the
code ("things that change together, stay together") or if a separate
README.md would be wise.

Off topic, do you recommend an ESP32 developer platform? Given it's
multicore it would be useful for debugging - I was nosing around Aliexpress
earlier, and a few caught my eye.

RobTillaart, 2025-01-21:
I have a few ESP32's mostly "ESP32 Wi-Fi and Bluetooth Board - CP2102" which I use for testing. 

Did a project once where one core controlled the (SPI based) UI and the other core moved a stepper motor, controlled timing and more. The display was rather large and assigning one core to it kept the updates smooth.

What might be interesting to do is a wokwi - https://wokwi.com/projects/389282282876510209  (it is just an example)

RobTillaart, 2025-01-21:
- https://wokwi.com/projects/389301981869402113   ESP32 + DS18B20

closed 2025-01-22 by milesburton