milesburton/Arduino-Temperature-Control-Library #274: 3.9.x vs 4.0.x

artofit, 2025-02-01:
Hi,

Thanks for updating the library.

Would you be so kind to state the benefits of the changes, in particular,
I was wondering the reason/usage for retryCount in getTemp or getTempC and why not in getTempF, seems odd to have it for Celcius and not Fahrenheit.

Thx

P.S.
Wouldn't it be good that some functions are inline?
In particular, 1 line functions, such as
such as getTempC(), getDeviceCount(), getResolution(), setAlarmHandler(), etc.

milesburton, 2025-02-04:
Hey @artofit,

Sorry for the delay.

The following PR (or the release notes on the 4.x tag) should answer why the change was made: https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/264

You do make a very good observation regarding the retry on the Fahrenheit. Frankly it was probably just an oversight.

To your point regarding inlining functions, yeah that could be a good change.

We've very open to any contributions from the community. If there's any improvements you believe could be made did you want to fork the repo and raise a PR for us to review?

milesburton, 2025-02-06:
I'll close this for now, but feel free to reopen if you'd like to associate a PR with the issue

closed 2025-02-06 by milesburton