milesburton/Arduino-Temperature-Control-Library #276: Always show "-127" on OLED.

mianqi2016, 2025-04-01:
I test this library on Arduino Nano, when I connected the data pin of DS18B20 to A7(D21), it would show "-127" in Serial Monitor by "Serial.print();", when changed A7 to D4, the showing is correct. When I want to print the temperature on OLED by "oled.print();" from [this library](https://github.com/mianqi2016/SSD1306Ascii), it always show "-127" either D21 or D4, D5, A0 ... why?

LRDPRDX, 2025-05-03:
Could you please provide the full code of your sketch when you get the "-127" when connected to `D21`. This value means your sensor is not connected. And more over from the documentation https://docs.arduino.cc/language-reference/en/functions/digital-io/digitalwrite/#_notes_and_warnings:

> The analog input pins can be used as digital pins, referred to as A0, A1, etc. The exception is the Arduino Nano, Pro Mini, and Mini’s A6 and A7 pins, which can only be used as analog inputs.

popos123, 2025-05-07:
As above, You need to provide the full code to the e.g. pastebin, it could be the problem with timing.
The problem with pin naming i have the same on STM32F401, the name "PB12" is working perfect, but "PB_12" is not.