milesburton/Arduino-Temperature-Control-Library #277: Readme Typo

ghoople, 2025-04-24:
In the Readme there's a typo "Connect a **4k7 kΩ** pull-up resistor between the 1-Wire data line and 5V power."

I presume this should be 4.7 kΩ based on the data sheet. 

Thanks for this great library!

RobTillaart, 2025-04-24:
Fyi
4k7 is the same as 4.7kΩ
This notation is sometimes sed e.g on PCBs where is not much space. So although it might not be the formal way it is not uncommon.

LRDPRDX, 2025-05-04:
@RobTillaart  I don't know if you just mentioned there is such a notation or meant there is no typo, but the typo is still there. Because it contains __two k's__ which can be interpreted as 4.7 kkOhm which is Mega :)

RobTillaart, 2025-05-04:
Sorry missed that second  k.
So indeed a typo, one way or the other.
Propose to change it to 4.7 kΩ which is less confusing.

milesburton, 2025-05-04:
> ID: <milesburton/Arduino-Temperature-Control-Library/issues/277/2849039266
> @github.com>

Indeed. Thanks both I’ll raise a PR

popos123, 2025-05-07:
Yes, a little typo, and i was noticed that the 4.7 kΩ is the minimum, sometimes doesn't work, so the good option is to add notice that the 2.2 kΩ or less is recommended (even on long distance e.g. 10 m is better 1 kΩ).

closed 2025-06-14 by milesburton