milesburton/Arduino-Temperature-Control-Library #281: 4.0.4 incompatible

MarcEngrie, 2025-07-27:
Hi,

I am running Platformio on VS Code. In one of my ESP32 project I am using your library "DallasTemperature". The version I am using is 3.11.0 (on platform Espressif 32 version 6.11.0). 

In the meanwhile you updated the lib to version 4.0.4, But Platformio tells me that an update to 4.0.4 is incompatible.

Can you tell me why or what I need to do to make it compatible again and able to upgrade now (and later on)?

[image]

milesburton, 2025-07-27:
Hmm, that's curious. You can stick on the current version, 4.0.4 included some minor security fixes - no functional changes. Let me explore the incompatibility issues that have been flagged up.

milesburton, 2025-09-02:
OK the reason is platformio is looking up via Github and I have yet to tag 4.0.4 so it's failing. I'll fix this shortly

milesburton, 2025-09-02:
That should work now, can you verify?

closed 2025-09-02 by milesburton

MarcEngrie, 2025-09-02:
just (Tue 02/09/2025 - 13:03 EST) tried and still the same. maybe I have to do something extra like ... ???

[image]

RobTillaart, 2025-09-02:
[image]

Seems updated on PlatformIO.
Shows mixed 4.0.4 and 4.0.5 labels - some may take more time to update (?).

milesburton, 2025-09-02:
> (milesburton/Arduino-Temperature-Control-Library#281)
>
> FYi: I tried to install the Dallas library on a project that has not
> already the library and also there I only see 4.0.4

As Rob mentioned, it may have not yet been propagated. I would suggest
using an older version. The last few versions are minor bug fixes and docs,
nothing ground breaking.